lowRISC/style-guides #1: [Verilog Style-Guide] Use 4 space indent only for wrapping long lines

sjgitty, 2019-07-02:
from @NilsGraf originally

**Suggestion**: Use 2 space indent everywhere (including module port declarations, module parameter lists, structs and enums). Use 4 space indent only for wrapping long lines.

**Rational**
There seems to be a mismatch between our Verilog style guide and the referred to C++ style guide. For example, the C++ style guide uses 2 spaces for structs and enum declarations, see this example:

```
struct UrlTableProperties {
  string name;
  int num_entries;
  static Pool<UrlTableProperties>* pool;
};
```
And this example:

```
enum UrlTableErrors {
  kOK = 0,
  kErrorOutOfMemory,
  kErrorMalformedInput,
};
```

However, our style-guide uses 4 spaces for enums, see here:
```
typedef enum logic [1:0] {  // A 2-bit enumerated type
    AccWrite,
    AccRead,
    AccPause
} access_e;
```

sjgitty, 2019-07-02:
I put my vote for 2 almost-everywhere.

I believe the only exception is truly long lines, like

```
assign extremely_long_variable_name =
    long_variable_name1 | long_variable_name2 |
    (long_variable_name3 & long_variable_name4);
```

but not these (shown as the preferred 2 spaces) plus the typedef shown above.

```
module modulename #(
  parameter1 = 100,
  parameter2 = 200
) (
  input [31:0] input1,
  ...
```

and

```
modulename #(
  .parameter1(128),
  .parameter2(256)
) instname (
  .input1( input1_i[31:0] ),
  ...
```

I believe that's what we're leaning towards.

closed 2019-07-03 by sjgitty

asb, 2019-07-03:
For those following along at home, there was a fair bit of discussion about this change over a period of time prior to the style guide being released publicly. There are may be one or two similar pending issues to be transferred over here, but following that issues will be discussed primarily on this repo issue tracker, with plenty of time for feedback etc.

PR #2 (2019-07-03): [verilog] convert coding style recommendation to two spaces
Fixes #1 

Convert the coding style recommendation to only use 4 spaces for true line continuation, 2 spaces in all other situations.