lowRISC/style-guides #4: [SystemVerilog] Macro naming convention

sriyerg, 2019-08-23:
The current Verilog style guide mandates ALL_CAPS for macro definitions:
https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#macro-definitions

IMO SV language already provides a semantic separation for macros through a preceding backtick. There doesn't seem to be a need to mimic C/C++ here. Most editors have SV syntax support so macros can be made out quite easily. 

DV code has heavy use of macro calls defined in UVM library, and all of them are lower_snake_case. UVM follows the following convention:

Macros representing constants: ALL_CAPS
```
`define FOO_WIDTH 32
```

Macros used for conditional compile: ALL_CAPS
```
`ifdef GATESIM
  // compile GLS specific code
`endif
```

Macros used for shortening a snippet of code: lower_snake_case
```
`define get_parity(val, odd=0) (^val ^ odd)

  p = `get_parity(rdata, cfg.use_odd_parity);
```

If we are mandated to use ALL_CAPS, then DV code ends up becoming a mix of macros invocations from UVM, which are lowercase and macro invocations from our libraries, in ALL_CAPS. This ends up making the code look ugly and inconsistent. 

I am inclined to follow the same convention as UVM, at least for DV. Seeking thoughts and opinions on this.

asb, 2019-08-27:
CC @sjgitty @NilsGraf @imphil @vogelpi @tomroberts-lowrisc for potential input here.

I can certainly see the problem with mixed styles. In LLVM we allow lower_snake_case for "STL-style" data structures or helpers for STL structures, to minimise the cases where there is a jarring switch between naming styles. We could imagine allowing something similar for DV and "UVM-like macros", even if we didn't encourage/allow this everywhere.

tomroberts-lowrisc, 2019-08-27:
Sounds sensible to allow it - should only really be an issue for DV since there shouldn't really be any macros in design RTL.

sjgitty, 2019-08-28:
> On a side note: using macros for shortening a snippet of code is perhaps
> not the most elegant way. The above example can lead to bugs if you call
> get_parity with an expression such as rdata & rdata_mask instead of just
> rdata, which then gets expanded to ^rdata & rdata_mask, instead of ^(rdata
> & rdata_mask) (so better to use parentheses around val and odd).
> So I think we should avoid using macros as much as possible, and keeping
> them all-caps can help making them less popular.
> It's also simpler to say in the style guide that *all* macros should be
> all-caps (instead of saying that some are all-caps, and some are

On this particular example, the burden is typically put on the macro writer
to ensure the user
can't make that kind of mistake.

   `define get_parity(val, odd=0) (^(val) ^ (odd))

On this particular issue (lower_case `defines for DV) I feel it is OK to
have a slightly different rule set for
DV in this case since there is already precedent with `uvm_error,
`uvm_info, etc.

On a related note, we currently have no guidance for style for DV. Should
we create a subsection
of the style guide for that? Should I file an issue in the style-guides
area @asb?

asb, 2019-08-28:
We'd definitely be open to DV specific style guidance if justified, so please do file issues for DV-specific exceptions (and/or a DV style guidance tracking issue).

sriyerg, 2019-08-29:
Thanks Nils for exposing the issue in the get_parity macro. 
+1 for adding DV style guide, since it goes beyond what is supported for RTL.

sjgitty, 2020-07-18:
@sriyerg @asb can we consider this issue closed with the addition of the DV style guide? @udinator

sriyerg, 2020-07-18:
Yes, this can be closed.

closed 2020-07-18 by sriyerg