lowRISC/style-guides #8: Discourage constant expressions in case statements

imphil, 2019-10-02:
`case (1'b1)` statements can be used as a replacement for a series of if/else blocks. It is essentially a style preference for some coders, but has the significant downside of miscompiling in Quartus Lite/Standard (and possibly other tools). Can we disallow or discourage its use?

See 
https://github.com/MarekPikula/quartus-sv-gotchas/blob/master/Intel%20Quartus%20SystemVerilog%20gotchas.md#1252-constant-expression-in-case-statement

and https://github.com/lowRISC/ibex/pull/336/files for more details.

Bad:

```verilog
unique case (1'b1)
  csr_save_if_i: begin
    exception_pc = pc_if_i;
  end
  csr_save_id_i: begin
    exception_pc = pc_id_i;
  end
  default:;
endcase
```

Good:

```verilog
if (csr_save_if_i) begin
  exception_pc = pc_if_i;
end if (csr_save_id_i) begin
  exception_pc = pc_id_i;
end
```

@sjgitty @eunchan @sriyerg @NilsGraf Any strong feelings/thoughts?

eunchan, 2019-10-02:
I've been used that coding style for ten years and had no issues on Xilinx Vivado, Cadence Palladium XP, SNPN Veloce, Synplify, and all EDA tools (VCS, Incisive, DC, Conformal, etc...). Should we discourage this only for a outdated tool? Or,, is it still an issue on Quartus II ?

The example above, Do we need `unique` in front of the `case`? I also put that in rv_plic `addr_hit` but it looks like I should avoid it as it isn't unique case anymore  just `case`.

imphil, 2019-10-02:
It's not about the unique, it's about the constant expression.

This problem exists for the latest version of Quartus Prime (aka Quartus II before the Intel acquisition of Altera) Lite and Standard, i.e. the version that must be used for all smaller and older FPGAs Intel produces. See https://github.com/lowRISC/ibex/issues/117 for a full discussion and https://www.intel.com/content/www/us/en/software/programmable/quartus-prime/download.html for a device compat matrix.

The situation is similar to ISE vs Vivado on the Xilinx side, with the additional fun of confusing naming and no freely downloadable version of the newer tool.

I'm not at all a fan of such workarounds, but since `case (1)` vs if blocks isn't really much more ugly either way I'd be willing to go for a syntax that happens to work on Quartus Lite/Standard, and enables many Intel users to use out code.

eunchan, 2019-10-02:
On Wed, Oct 02, 2019 at 09:50:56AM -0700, Philipp Wagner wrote:
> It's not about the unique, it's about the constant expression.
>
> This problem exists for the latest version of Quartus Prime (aka Quartus II before the Intel acquisition of Altera) Lite and Standard, [...]
One thing to add is.. This scheme `case (1'b1)` is somewhat famous for enforcing 
one-hot encoding for FSM. I am not sure tools like verdi can recognize it as FSM 
if we chance it to `if .. else if ...  else `

NilsGraf, 2019-10-02:
> ```verilog
> end if (csr_save_id_i) begin
> ```
Is this a typo and should read `else if` instead?

sriyerg, 2019-10-02:
The snippet of code indicated in the PR does not seem to be modeling an FSM. In that case it should be okay to change that to if-else (also, ok to ban such a practice, as long as it does not describe an FSM). 

When modeling an FSM though, case (1'b1) is perfectly valid way (hot bit FSM) and most tools that provide coverage collection capabilities detect it as such, so that should be allowed. 

Thanks
Sri

sjgitty, 2019-10-06:
Re the FSM comments, wouldn't you normally use
`case(fsm_cs)` instead of `case(1'b1)` in those cases?

Personally it is quite surprising that a tool doesn't accept `case(1'b1)`
since it has been around for decades. It also used to be the case that
parallel case(1'b1) was quite different than if/else if/else if. Is that not
longer the case? Should we recommend at least `unique if` in this
situation?

eunchan, 2019-10-07:
I am not using case (1'b1) for FSM. But I saw many modules using the style
for FSM to explicitly use one hot encoding.

MarekPikula, 2019-10-07:
I can recheck on some simple example if indeed it doesn't synthesize correctly with Quartus. Maybe in case of the code I analysed some other optimizations kicked in and the whole block was misinterpreted.

MarekPikula, 2019-10-16:
As pointed out in lowRISC/ibex#336 the constant expression case synthesizes correctly. It was a mistake on my side. Issue can be closed.

imphil, 2019-10-16:
Closing as there's no strong need for discouraging this syntax any more.