lowRISC/style-guides #13: Dealing with removing X in cases where consuming rather than producing X is the error

GregAC, 2019-12-09:
The current proposals for removing the use of X from RTL (https://github.com/lowRISC/style-guides/pull/12) focusses on cases where the generation of X would have been the error, so replacing the use of X with appropriate assertions and coding style changes is a sufficient replacement.

There are also areas where the X generation itself isn't because of an exceptional condition but rather flags a particular wire should have it's value ignored. So using the generated X value is the error you want to trap.

A typical example of this is on a bus interface. A valid signal accompanies some request information (address, type etc). When the valid signal is low the request information shouldn't be used and can be driven to X to signify this.

An example of a bug this can find is as follows (this could be a code fragment from a device acting on requests from the bus, NEVER_X is a macro that gives you an assert that fires if the specified signal becomes X)

```verilog
...
input logic         req_valid_i
input logic [7:0] req_addr_i
...

assign action_addr = req_addr_i == `ADDR_ACTION;

// Bug here, should have & req_valid_i
assign action_en = action_addr_i;

`NEVER_X(action_en);
```

When req_addr_i is driven to X when req_valid_i is low the assertion will quickly fire and the bug found an fixed. Where X is not used and the RTL doesn't tend to generate (ADDR_ACTION) as a value for req_addr_i when req_valid_i is low.

We can have explicit X driving in the DV environment on specific buses to help recreate this but you'd still want the NEVER_X macro or similar to ensure you get a quick, easy to diagnose failure. Plus you may want similar checking on internal signals that aren't well defined interfaces the DV can probe/drive values on (e.g. in Ibex we could set the write address of the register file to X when we're not writing a register).

One replacement would be for ``` `NEVER_X(action_en)``` is be replaced by an assertion ```action_en -> req_valid_i```. In the style guide we would recommend any 'enabling' signal that triggers an action has an assertion that checks if it is set the appropriate 'upstream enabling' signal is also set.

You can flip the above example on its head and instead have a bug where req_valid_i is being asserted when there is no valid request only this time we've fixed the bug above.

```verilog
assign action_en = action_addr_i & req_valid_i;
```

If the RTL generating the address drives X on address when it's not intended to produce a request (but the RTL that generates the valid signal has a bug and sets valid anyway) again we'll quickly hit an easy to diagnose failure. Without the X if ```req_addr_i``` is mostly 0 when no request is meant to be generated (and a request against address 0 is a no-op) but occasionally is ``` `ADDR_ACTION``` when no request is meant to be generated we've got a failure that we're less likely to see (and my be harder to diagnose when we see it) for the same bug.

I can't immediately think of a good replacement for this (whether that's an assertion or a coding style change). Though perhaps a bit of an artificial example we're not hugely worried about in practice?

msfschaffner, 2019-12-09:
Thanks for filing this issue, @GregAC.

I think it would be fine to allow the DV environment to drive X on bus models and require the associated RTL modules to have `` `NEVER_X`` (== `` `ASSERT_KNOWN``) assertions on their inputs. But as you mentioned, this would only be a partial solution since everything embedded in or driven from the RTL side is not allowed to leverage X assignments according to #12. 

Not sure whether it would be a good idea, but if we are mainly concerned about testing bus functionality here, would it help if we used specific known bad addresses to flag the type of errors you are worried about? E.g. something like `0xDEAD_BEEF` or `0xFFFF_FFFF` that we could trigger on using an assertion?

Your second example feels a little bit artificial to me... i.e. if you assume that X will always be correctly assigned in the code, is it reasonable to assume that if you assigned another value like 0 (instead of X), the address can now suddenly be assigned `ADDR_ACTION` occasionally?

weicaiyang, 2019-12-10:
> When req_addr_i is driven to X when req_valid_i is low the assertion will quickly fire and the bug found an fixed. Where X is not used and the RTL doesn't tend to generate (ADDR_ACTION) as a value [...]

I made a [PR](https://github.com/lowRISC/opentitan/pull/1134/files) to enforce that driver must set data/address/control signals to be X when valid==0, so this kind of issue should be caught by DV.
```systemverilog
  function void invalidate_a_channel();
    vif.host_cb.h2d.a_opcode  <= tlul_pkg::tl_a_op_e'('x);
    vif.host_cb.h2d.a_param   <= '{default:'x};
    vif.host_cb.h2d.a_size    <= '{default:'x};
    vif.host_cb.h2d.a_source  <= '{default:'x};
    vif.host_cb.h2d.a_address <= '{default:'x};
    vif.host_cb.h2d.a_mask    <= '{default:'x};
    vif.host_cb.h2d.a_data    <= '{default:'x};
    vif.host_cb.h2d.a_user    <= '{default:'x};
    vif.host_cb.h2d.a_valid   <= 1'b0;
  endfunction : invalidate_a_channel
```

GregAC, 2019-12-18:
I hadn't initially appreciated we'd be having liberal use of X check assertions to catch cases where X accidentally is created. That along with the PR from @weicaiyang should get us most of the of the way there.

Perhaps in the style guide add something that says for internal signals that could exhibit this kind of issue use some value likely to trigger a failure if it was accidentally consumed (e.g. for the ibex decoder default to RF we=0 addr=1 so if we accidentally write we'll clobber a register and see a failure), though this could conflict with security requirements (we can just glitch we to get a spurious write now rather than we and addr). Using the initial assertions I suggested may be best (i.e. if you have a signal that triggers an action consider adding an assertion that checks some primary enable or trigger that must be set for the action to occur is set). I can have a go at a PR against the style guide and see what people think.

msfschaffner, 2019-12-18:
I am leaning towards using the assertions you suggested since in that case we do not have to significantly alter the RTL behavior...

do you want to make a suggestion once https://github.com/lowRISC/style-guides/pull/12 has made it?

vogelpi, 2019-12-19:
I think this is a good use case for assertions and we should in general make more use of SVAs. Changing the RTL to likely trigger a failure at the consumer end does not sound like a good idea to me. From the RTL itself, the intent of such a statement may not always be obvious. If at some point the consumer is changed and no longer triggers a failure the issue at the producer can be masked. In contrast, the intent of assertions is always clear and they are more tied to the producer.

I am also in favor of using assertions for this instead of changing the RTL. Having some guidance on how and when to put SVAs in the RTL is a good idea.

msfschaffner, 2020-02-12:
pinging this thread :). @GregAC maybe this can be consolidated with style guide changes from #16 ?

GregAC, 2020-02-25:
Addressed by #27

closed 2020-02-25 by GregAC