lowRISC/style-guides #14: [security] Provide RTL coding style guidance in case of security critical applications

msfschaffner, 2020-01-02:
We should add a separate RTL style guide for security related applications. The actual contents of that style guide still need to be fleshed out, but some aspects that should be covered have surfaced in #12 (e.g., guidance on FSM coding and placing of security-related assertions).

cdgori, 2020-04-06:
PR1877 triggered something in me that needs to be incorporated here:

We might need to incorporate some "blanking" muxes to prevent partial intermediate values from propagating to arithmetic functions/operators.  Especially if we have two arithmetic operators in sequence in a single clock cycle - e.g. y = g(f(x)), you really don't want partial f(x) fed to g() so you need to have some "done" or "valid" indicator from f(x) to allow g() to process the output of f(x).  Pipeline registers serve the same function here, but sometimes you are trying to minimize latency.

For public-key crypto (bignum comes to mind), I think we also are concerned about glitching where part way through a clock cycle some sub-words/bytes/nibbles have been computed but others are not yet ready, and a clock glitch could "jump ahead" in the algorithm - I don't have a concrete example here though and will have to discuss with @felixmiller / others to see if there is a real concern.  (And what techniques we should use to protect.)

tjaychen, 2020-09-24:
should we link the WIP doc to this issue?

sjgitty, 2020-09-25:
If you're referring to @cgori's doc, that is not ready yet. This old
response (nice dig up, @tjaychen), should be incorporated into that one,
since this is a good idea. I imagine said blanking muxes should be more
like random XORs to avoid hamming concerns?