lowRISC/style-guides #16: Dealing with 'unavoidable' Xs to case statement inputs

GregAC, 2020-01-14:
This Ibex issue: https://github.com/lowRISC/ibex/issues/548 has raised the question of how we should deal with Xs going into case statements where we cannot avoid that X. From the style guide a typical case statement should look something like

```verilog
`ASSERT_KNOWN(CaseInputKnown, case_input, clk_i, !rst_ni)
always_comb begin
  case_output = 4'b0000;
  case(case_input):
    2'b00: case_output = 4'b0001;
    2'b01: case_output = 4'b0010;
    2'b10: case_output = 4'b0100;
    2'b11: case_output = 4'b1000;
    default: ;
  endcase
end
```

How should we deal with the case where `case_input` could be X? e.g. it could come direct from a top-level input that the DV environment may drive to X or from the output of a memory.

In such cases there's generally some kind of valid signal (`case_valid`) that lets us know that `case_input` can be ignored and hence `case_output` should be ignored.

You could say `case_input` must be qualified by `case_valid` to remove the X from the case input

```verilog
logic [1:0] case_input_qualified;

assign case_input_qualified = case_input & {2{case_valid}};
`ASSERT_KNOWN(CaseInputKnown, case_input_qualified, clk_i, !rst_ni)
always_comb begin
  case_output = 4'b0000;
  case(case_input_qualified):
    2'b00: case_output = 4'b0001;
    2'b01: case_output = 4'b0010;
    2'b10: case_output = 4'b0100;
    2'b11: case_output = 4'b1000;
    default: ;
  endcase
end
```

Though this can lead to timing issues. In the case in the linked Ibex issue the valid signal (which is whether or not the instruction is illegal) is fairly late signal which we don't want to factor into the decoder outputs that lead to the X going into a case statement input.

Another option is to allow the valid signal to qualify the assertion:

```verilog
`ASSERT(CaseInputKnown, case_valid |-> !$isunknown(case_input), clk_i, !rst_ni)
always_comb begin
  case_output = 4'b0000;
  case(case_input):
    2'b00: case_output = 4'b0001;
    2'b01: case_output = 4'b0010;
    2'b10: case_output = 4'b0100;
    2'b11: case_output = 4'b1000;
    default: ;
  endcase
end
```

I did also wonder if we could use assertions on the case output instead:

```verilog
`ASSERT_KNOWN(CaseOutputKnown, case_output), clk_i, !rst_ni)
always_comb begin
  case_output = 4'b0000;
  case(case_input):
    2'b00: case_output = 4'b0001;
    2'b01: case_output = 4'b0010;
    2'b10: case_output = 4'b0100;
    2'b11: case_output = 4'b1000;
    default: ;
  endcase
end
```

Though given the coding style already says you should avoid explicit Xs I don't think this really gives us anything.

Really this comes down to what is the ASSERT_KNOWN really for? As we also include a default statement we don't have the simulation/synthesis mismatch issue where a simulated X gives us latching behaviour and hence want to stop that from occurring in simulation. So I see it more as a useful flag showing we've got some X's in the design that need investigating in which case qualifying the assertion with the valid signal seems reasonable.

GregAC, 2020-01-14:
@msfschaffner @sjgitty any thoughts?

msfschaffner, 2020-01-14:
Thanks for bringing up this issue. I would go for your second solution (qualified assertion).

Our guidelines state that such assertions on case select signals are strongly *encouraged* and that the style of this assertion is up to the designer. It may range from simple KNOWN assertions up to functional assertions. So in my view, an assertion qualified with a valid signal is permissible here.

While I would not place the KNOWN assertion on the case statement output in this case (there is not really a point in doing that if you correctly assign a default), there should be KNOWN assertions on the primary outputs of the block as these are mandated per style guide.

In the end we want to make sure that RTL does not latch any X values and propagate them to logic downstream. In this case, the qualified assertion at the input basically checks whether the input data is valid under normal operation. The KNOWN assertion at the output of the module is a safeguard that would flag a case where Xes are latched and propagated.

sjgitty, 2020-01-16:
+1 to the qualified assertion, with a comment as to why it is qualified
and not just a standard `ASSERT_KNOWN`.

msfschaffner, 2020-02-12:
@GregAC you have added qualified assertions to `prim_assert.sv`, right? 
do you want to take a stab at adapting the style guide accordingly, such that we can close down this issue?

GregAC, 2020-02-13:
Yup I'm in the process of doing some updates to address this and #13 just had some other higher priority stuff to work on that has delayed it.

GregAC, 2020-02-25:
Addressed by #27

closed 2020-02-25 by GregAC