lowRISC/style-guides #18: Remove case-inside construct restrictions in our style guide

msfschaffner, 2020-01-18:
Carried over from https://github.com/lowRISC/opentitan/issues/901

I think it is time to revisit whether case-inside is supported by the tools, and remove that restriction from the style guide if it is, see
https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#wildcards-in-case-items

The case-inside statement can come in pretty handy for address decoding, and afaik, the debug module uses it, which also means that all our tools we've used so far are able to cope with it:
https://github.com/pulp-platform/riscv-dbg/blob/811b2d707795a5044d63a68f747b2f9cd29d3a88/src/dm_mem.sv#L207-L212

vogelpi, 2020-01-20:
Thanks @msfschaffner for moving the issue over.

Indeed, the tools we have been using so far support it. I would agree to allow it.

However, also for this case there are the usual tools which do not support it such as [Quartus](https://www.intel.com/content/www/us/en/programmable/quartushelp/current/index.htm#hdl/vlog/vlog_list_sys_vlog.htm). I don't know if being 100% compatible with Quartus is required. Also, some versions of this tool have already other incompatibilities with our style guide, see also lowRISC/ibex#117.

msfschaffner, 2020-01-21:
@vogelpi, do you know whether the Quartus issues in that Ibex thread have ever been reported to Intel?

vogelpi, 2020-01-21:
If I remember correctly, most of the issues discussed in the Ibex thread affects a version of Quartus that is no longer supported by Intel, but that is the only option for some older/low-end FPGAs (besides third-party synthesis tools). Thus it probably does not make sense to report this to Intel.

What is more of a problem for this issue here the other link pointing to the documentation of the latest Quartus Prime Pro version (19.4 released in Dec 2019). It seems as if it only supports SystemVerilog 2005 and 2009. The case inside construct is only available as of SystemVerilog 2012.

msfschaffner, 2020-01-21:
Ok, I see. In that case the question boils down to whether we want to support Quartus Prime Pro officially. 

If we leave the style guide as is, we would have to clean up all these statements inside the RISC-V debug module at some point, because that won't work with Quartus...

GregAC, 2020-01-21:
re: Quartus compatibility we could use the sv2v tool (https://github.com/zachjs/sv2v) which we're already using for yosys which converts system verilog to verilog.

vogelpi, 2020-01-27:
> If we leave the style guide as is, we would have to clean up all these statements inside the RISC-V debug module at some point, because that won't work with Quartus...

Since the RISC-V debug module is an upstream IP, we can also create a patch to change these statements for OT. But I am not convinced this is the way to go.

Personally, I think we should start using this language feature (it's 2020 and this was added in 2012). Referring people relying on Quartus to sv2v as suggested by @GregAC could indeed work. There might be even more language constructs allowed by our style guide that the newest version of Quartus does not support. I have used sv2v as well, but only for small stuff.

Maybe also @imphil could share his view.

imphil, 2020-01-27:
Disallowing `case inside` was something that we had in the style guide from the start, probably being carried over from the Google-internal style guide. I don't know what tools it referred to. @sjgitty or @msfschaffner can you look at the current Google-internal style guide, and/or the changes to it to find the motivation for adding this restriction in the first place?

We certainly didn't add the restriction in response to a Quartus problem (which doesn't mean Quartus supports `case inside`, its just untested at this point).

sjgitty, 2020-01-27:
I'm checking the history on this.

sjgitty, 2020-01-27:
The problematic tool was Vivado 2016.3. I will work on this side to see if
it
has been fixed or not in the latest Vivado version.

sjgitty, 2020-01-27:
@eunchan has run the reported problem cases against the recent
Vivado tools and reported that they work fine now. For this particular
issue, we're fine with removing the 'case inside' restriction. There is
no other known reason for the restriction on our side.

msfschaffner, 2020-01-30:
I have checked the Vivado release notes a while ago and this has apparently
been fixed in 2017.3.
We have been using the DM with those constructs successfully also in all
the Ariane projects since 2018.
See: https://www.xilinx.com/support/answers/64777.html

imphil, 2020-01-30:
Thanks @msfschaffner. I'm OK with removing this section then, please open a PR.

closed 2020-02-12 by msfschaffner