lowRISC/style-guides #19: Remove restriction of wildcard `import my_pkg::*;` inside the same module

eunchan, 2020-01-21:
I would like to remove the clause that restricts the usage of wildcard import of the package.

In some cases, it is quite useful not to type the package namespace every time.
The concerns causing the restriction above is the collision of the names between the module and the package and the usage of wildcard import outside of the module. But if the wildcard import is used inside the module and the same IP that defines the package, it is unlikely to have collision.

So I would like to suggest below cases are permitted in the style guide.

1. wildcard import at the module header is permitted if the package is defined in the same IP.

```systemverilog

module mod_a import mod_a_pkg::*; (
);

endmodule
```
2. wildcard import is permitted inside the module body if the package is defined in the same IP.

```systemverilog
module mod_a ();

  import mod_a_pkg::*;

endmodule
```

Other than the cases above, wildcard import is not allowed. For instance, below example can create name collision in the module following mod_a in the source list.

```systemverilog
import mod_a_pkg::*;

module mod_a ();
```

Other bad examples:

```systemverilog
// wildcard import for other packages outside of the IP
module mod_a import mod_b_pkg::*; ();
```

```systemverilog
module mod_a ();

  // wildcard import of different packages.
  import mod_b_pkg::*;

endmodule
```

**directory structure for above examples**

```console
hw/ip
    | mod_b/rtl/mod_b_pkg.sv
    \ mod_a/rtl/
        | mod_a.sv
        \ mod_a_pkg.sv
```

CC: @tjaychen @msfschaffner

msfschaffner, 2020-01-21:
Indeed, I support this proposal. This issue has in part also been brought up by @mwbranstad (we just did not yet file it as an issue on GH).

mwbranstad, 2020-01-21:
Agreed. My original comment to @msfschaffner was:
•	Wildcard packages: Wildcard import (of packages), eg. import my_pkg::*;.
o	This style is being commonly used for registers, so this statement is a bit confusing.

GregAC, 2020-01-22:
We do already have examples of 2. in Ibex so if we're deciding not to allow this we have some cleaning up to do!

I'm in favor of this with the rules @eunchan specifies.

tomroberts-lowrisc, 2020-01-22:
+1 from me

asb, 2020-01-22:
I've got no objection either. I think this is good to go to a PR. Will you be following up with one @eunchan?

eunchan, 2020-01-22:
Yeap. I will create a PR to modify the sentence.

GregAC, 2020-02-07:
This has been fixed by https://github.com/lowRISC/style-guides/pull/20

closed 2020-02-07 by GregAC