lowRISC/style-guides #24: [verilog stlye guidelines] parameters should be all caps

mwbranstad, 2020-02-12:
I have noticed that verilog passed parameters are recommended to be coded as camelcase style. I think this is bad idea. My only argument is that all capital letters is the defacto standard in the industry. Every style guideline that i have every read or created states this. 

It is clear that this style is ubiquitous. I notice that under the heading of Parameterized Types in the verilog style guidelines, there are examples showing all capitial letter parameters there. Even if this issue is rejected, some work needs to be done to clean up the style guidelines in its present state.

Additionally, I believe this applies to both local parameters and emul types.

msfschaffner, 2020-02-12:
related to https://github.com/lowRISC/ibex/issues/579#issuecomment-584379063

mwbranstad, 2020-02-12:
> Personally, I think using PascalCase (UpperCamelCase in the style guide) in the tunable parameter is good even though it is slightly off from the industrial standard. the borderline isn't that obvious between what the tunable parameter is and what constant is, but at least the different form of the word helps me to distinguish.
> 
I am trying to think of case where a parameter is not "tunable". I am guessing that constants are the case where you want to pass a non-tunable, but trying to figure out why one would do this. Anyway, thanks for the explanation and history.

sjgitty, 2020-02-14:
(Similar apology for slowness of response. Thanks @mwbranstad for bringing these up.)

Oh boy, this is a bit uglier than I thought. https://github.com/lowRISC/stwg-base/issues/427 raised an issue, solved it in https://github.com/lowRISC/stwg-base/pull/749, then left this note "This will next be rolled into the already-public lowRISC style guide". But that never happened. :scream:  (Sorry for the pre-history references).

This means that our still-dangling-for-no-good-reason `doc/rm/verilog_style_guide.md`
in lowrisc/opentitan is *newer than this one*! Not a good state of affairs.

Let's do this:
1. @mwbranstad read through the style guide in opentitan, see if it is better or worse than what is here w.r.t. your concerns. I don't think it "solves" them. Update this issue with the thoughts w.r.t. that.
2. @sjgitty will rectify the discrepancy of the two style guides ASAP with a PR here
3. @sjgitty will remove the opentitan VSG since we've been wanting to for a while.
4. We need an answer on https://github.com/lowRISC/opentitan/issues/988, which is how/if to render

My apologies for letting it get to this state of affairs.

imphil, 2020-03-12:
https://github.com/lowRISC/opentitan/pull/1733 removed the outdated style guide from the opentitan repository, finishing the last outstanding items in this issue. Thanks @mwbranstad for your report, and sorry it took so long to completely resolve it. Please let us know if there's anything left to do from your side!

mwbranstad, 2020-03-16:
Reopening issue since it appears to only partially address the issue. After reading the verilog guidelines as pointed to by https://github.com/lowRISC/opentitan/pull/1733, in general found that most of the module parameter references were fixed to be camelcase, now being consistent with the parameter preference. But similar to the enumeration description, I was hoping for both a preferred form (camelcase) and allowed form (all caps) option. Additionally, the state machine example (a good example) should have the preferred form (all caps) rather than the allowed form (camelcase).

imphil, 2020-03-16:
@mwbranstad would you mind creating a pull request to the style guide implementing your suggested guidance? It's probably easier to find agreement if we have the specific text in front of us.

mwbranstad, 2020-03-16:
@imphil yes, I will do that.

mwbranstad, 2020-03-23:
@imphil I was trying to find the VerilogCodingStyle.md in the OT repo, but it must be somewhere else. Can you help me find it to prepare it for a PR?

sjgitty, 2020-03-23:
Mark, it was confusingly in the OT branch too, but the real copy is here:

https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md

mwbranstad, 2020-03-31:
Now addressed in PR:
https://github.com/lowRISC/style-guides/pull/32
Will close once merged.

rasmus-madsen, 2020-06-30:
@mwbranstad  can this be closed?

mwbranstad, 2020-06-30:
@rasmus-madsen yes it can. Discussions related to this topic have resulting in no change to the style guidelines.

closed 2020-06-30 by mwbranstad