lowRISC/style-guides #30: [UVM:styleguide] best end of test practices

rasmus-madsen, 2020-03-05:
from #22 a discussion on where and how to use objections has been started.
to not clutter that PR too much I have taken this into its own issue.

The original suggestion is to avoid objections in drivers and monitors and instead add them in the scoreboard.

I will try to make an argument why monitors is the perfect place for objections and why they should only go into phase_ready_to_end() 

one of the Arguments of why it is nice to have objections in the scoreboard is given by @sriyerg 
I took the liberty of quoting it here:

> I see some valid usecases in allowing raising/dropping objections in the scoreboard; example: in xbar type DUT, raise obj when transaction enters the DUT and drop only when it exits. This objection (that helps ensure that the test does not exit prematurely) cannot be captured in the monitor and has to be captured in a higher level component. It cannot be captured in the test either since it only generates and sends C-R stimulus to the driver. It has no knowledge of when the transactions actually enter and exit the DUT.
> 
> We keep the performance impact low by raising objection exactly once (dont raise if it is already raised), which is what the line implies. You can find that logic in our base scoreboard (`hw/dv/sv/dv_lib/dv_base_scoreboard.sv`). If by deadlocks you mean hangs, then there is an obvious problem either in the design or dv, that needs fixing. We avoid hanging forever through reasonable test timeouts.

I have two arguments against this - the first one being simply performance.
every time you raise and drop an objection is causing the component hierarchy to be traversed.
so for shallow environments this might be a minor impact but as the hierarchy gets deeper this quickly gets expensive.

The other argument is that implementing this in the scoreboard requires a watchdog timer to be implemented a long the side to avoid deadlocks.

Here is what I suggest

Most modules have deterministic input to output latency and does not need any objections besides test case objections and a good choice of drain time. this should be the default go to.
objections should only be used outside the test when the DUT latency is not detministic!

Some modules does not have deterministic latency. an example can be some sort of packet builder where input packets are grouped in larger chunks and these chunks are only forwarded when reaching a certain size.

Modules like this usually have an internal timeout to flush out undersized chunks.

In such a scenario it will be hard to raise objections in a scoreboard as the scoreboard cannot easily predict the number of outputs based on the number of inputs.
and the suggested objection method would be insufficient.

instead I suggest putting the objetion inside the phase_ready_to_end() function.
this is called once all modules have dropped their run_phase objections and the test is trying to end the run phase.

here is what is currently in my scoreboard (AES) 

```
 virtual function void phase_ready_to_end(uvm_phase phase);
    if (phase.get_name() != "run") return;
    if ( (dut_fifo.num() != 0 ) || (ref_fifo.num() != 0) ) begin
      phase.raise_objection(this, "fifos still has data");
      fork begin
        wait_fifo_empty();
        phase.drop_objection(this);
      end
      join_none
    end
  endfunction

  virtual task wait_fifo_empty();
    wait ((dut_fifo.num() == 0 ) && (ref_fifo.num() == 0));
  endtask

  // ...
endclass
```

This ensures the Objection is ever only raised once!
obviously this still has the deadlock issue so a time would need to be implemented in parallel to the ''' wait_for_fifo_empty ''' function.

But this is still sub-optimal because the scoreboard should be a 0-time and there for it does not have any knowledge of clock frequency and one would have to guess what number of milliseconds is enough for any used clock frequency. 

I believe the better choice is to implement a function in the monitor that detects traffic and keeps the phase from ending for /# number of clocks.
this is very similar to drain time but where drain time is for the full environment this implementation will work even when  multiple environments are cascaded to form a top level verification environment. 

the idea is to start a timer when phase_ready_to_end is called.
if no traffic is seen before a timer runs out - it is ok to end.

here is an example I did

```
function void phase_ready_to_end(uvm_phase phase);
      uvm_phase current_phase;
      string str ="";
      uvm_object objectors[$];
      current_phase = phase;

      if (phase.is(uvm_main_phase::get())) begin       
      // Start watchdog
        fork
          watchdog_ok_to_end();
        join_none;
        if (!ok_to_end) begin
          phase.raise_objection(this);
          fork begin
            wait_for_ok_end();
            phase.drop_objection(this);

            // see who is still objecting
            current_phase.get_objection().get_objectors(objectors); // get all that has raised objections
            str = $sformatf("\n\n\t |--------------------| Who Is Still OBJECTNG |------------------------|"); 
            str = {str, $sformatf("\n\t  current phase: %s ", current_phase.get_name())};
            str = {str, $sformatf("\n\t Hierarchical Name                                              Class Type")};
            foreach(objectors[i]) begin
              str = {str, $sformatf("\n\t%-60s%s\n\n", objectors[i].get_full_name(), objectors[i].get_type_name())};
            end
            `uvm_info({LOG_NAME, "*"}, str, UVM_HIGH);
          end join_none;
        end
      end
    endfunction

    //-----------------------
    // Wait until ok_to_end is set. Only checks ok_to_end every timeout_delay cycles since this is the length of the watchdog cycle
    task wait_for_ok_end();
      forever begin
        repeat(timeout_delay) @(posedge vif.clk_i);
        if (ok_to_end) break;
      end
    endtask // wait_for_ok_end

 // Watchdog waits timeout_delay cycles while checking if there is any data on the interface. If there is then !ok_to_end else ok_to_end.
    task watchdog_ok_to_end();
      bit        reset_timer = 0;
      fork
        begin        
          forever begin         
            @(posedge vif.clk_i);
            if(vif.vif.rd) begin           
              reset_timer = 1;
            end          
          end
        end
        begin
          forever begin
            ok_to_end = 0;
            repeat(timeout_delay) @(posedge vif.clk_i); // stay here until #timeout delay clocks has passed
            if(!reset_timer) begin
              ok_to_end = 1;
              break;
            end else begin
              `uvm_info(LOG_NAME, $sformatf(" Resetting timer"), UVM_FULL);
              reset_timer = 0;
            end          
          end
        end        
      join_any
    endtask
```

I hope this at least shows some of the drawbacks of putting objections in the run phase.
and secondly in the scoreboard.
If there is a big desire to but objections in the scoreboard I think that is ok, but I think we should require objections to only go in the phase_ready_to_end function when outside of the test.

weicaiyang, 2020-03-06:
@rasmus-madsen thanks for bringing this up.

I didn't use `phase_ready_to_end` before. After do some study, I think it's a good idea.

With raise/drop objection in test and drain time, the other components just need to use `phase_ready_to_end` to make sure last transaction completes gracefully. This reduces a lot of raise/drop objection and simplify the objection logic as well.

I personally prefer the way you did in aes. I modified a little bit for reuse.
```
// this function can be put in base scoreboard (dv_base_scoreboard) and reused by the other IPs
 virtual function void phase_ready_to_end(uvm_phase phase);
    if (phase.get_name() != "run") return;
    // perhaps remove this IP specific code and raise this objection anyway
    //if ( (dut_fifo.num() != 0 ) || (ref_fifo.num() != 0) ) begin 
      phase.raise_objection(this, "fifos still has data");
      fork begin
        wait_fifo_empty();
        phase.drop_objection(this);
      end
      join_none
    //end
  endfunction

  // each IP should override this task 
  virtual task wait_fifo_empty();
   // # (drain_time); // wait for some drain time if necessary
    wait ((dut_fifo.num() == 0 ) && (ref_fifo.num() == 0));
  endtask
```

A scoreboard may have transactions from multiple agents, like xbar (10+ agents), uart (2 agents).
I feel we should raise objection in `phase_ready_to_end` in monitor when bus isn't idle. And this part can be easily reused cross project since it's in agent.
Then scoreboard can just use its available high-level info (like fifo size) to determine if it's ready to end.

I think I would try this in spi_device. Thanks again for bring this idea to opentitan.

rasmus-madsen, 2020-03-09:
thank you @weicaiyang I also find this solution very elegant.
I still think checking fifo fill levels is valid - but I think this should be done in the check_phase() at the end of test

I suggest we 

- recommend objections to be put in the monitor
- recommend that fifo fill levels in the scoreboard is checked during the check_phase (not in the ready to end phase)
- Require objections are put in phase ready to end.

this would still allow for objections in the scoreboard, but enforce that they are in the phase_ready_to_end() function.

@sriyerg @udinator 
what are your thoughts?

weicaiyang, 2020-03-09:
@rasmus-madsen can you elaborate what to do in phase_ready_to_end in monitor and scoreboard?
For monitor, raise objection at phase_ready_to_end when bus isn't idle?
For scoreboard, since fifo level is checked during check_phase and bus idle is checked in monitor, what would you suggest to do in phase_ready_to_end of scoreboard?

rasmus-madsen, 2020-03-09:
@weicaiyang 
for the monitor I am suggesting implementing a watchdog that is activated when the monitors phase_ready_to_end() function is called this watch dog raises and objection.
if a transaction is seen on the bus in before the watchdog times out the watchdog timer is reset.
else the watchdog will end and the monitor will drop the objection and allow the phase to end.

I am suggesting not putting anything in the ready_to_end_phase() for the scoreboard as it is simply not needed if the monitors are coded correctly and the drain time in the base test is set to something that resembles the maximum latency through the DUT.
the output monitor will catch any transactions that will be passed to the scoreboard - and since the scoreboard handles these in 0 time there will never be an actual need to wait for the scoreboard to finish.

The problem with having something in scoreboards phase_ready_to_end is that it would be waiting for something to finish i.e a fifo being emptied which it can ALWAYS lead to a deadlock if what you are waiting for never comes unless you have a timeout running in parallel.

in the monitor you are actually checking that nothing happens which will NEVER lead to a deadlock.

But I think we should allow objections in the scoreboard as long a we recommend not to.
and insist on adding a watchdog timer in parallel.

I can make some example code if needed

weicaiyang, 2020-03-10:
@rasmus-madsen thanks for the explanation.
It sounds good to me. Looks like we can implement most of codes in base monitor (dv_base_monitor) to handle this and all the monitors can reuse that.

rasmus-madsen, 2020-04-21:
The style guide still needs updating with the sections above.
While this may not be urgent - I consider it fairly important as the current content is suggesting methods that the suggestion above discourages.

udinator, 2020-04-24:
@rasmus-madsen - Something that was just recently brought up in a PR for the Icache testbench is the `uvm_heartbeat` mechanism (described [in this paper](http://www.synapse-da.com/Uploads/PDFFiles/04_UVM-Heartbeat.pdf)) to be used for EOT, which would have the side effect of removing the need for a global test timeout due to the watchdog nature of this mechanism.
I personally am not familiar with this approach, so just wanted to start a discussion about any potential benefits/drawbacks of this mechanism for EOT checking versus the proposed approach of using `phase_ready_to_end` in the monitors. What do you think?

rasmus-madsen, 2020-04-24:
Did you remove your comment again I don’t see in GIT?
I was not aware of his class

I only scanned the document briefly so I will need to do a deeper dive into it.
but It looks like this one looks for objections rather than transactions
Which I at least for now does not find very useful as I advocate for limiting the number of objections.
as these are very expensive due to the traversing of the hierarchy.

But thank you for the paper I will do some more exploration into the subject.

Rasmus Madsen
Technologist

Western Digital®
Email: rasmus.madsen@wdccom
Mobile: +45-3161-3969

rswarbrick, 2020-04-24:
I agree that the heartbeat mechanism isn't really related to end-of-test (unless something bad happens!), so probably isn't relevant here. I also got the email and can't see the message: maybe Udi deleted it when he realised the same?

udinator, 2020-04-24:
Ah sorry! I did indeed delete the comment when I realized that the heartbeat is not really an EOT mechanism; I'll re-send the [link to the paper](http://www.synapse-da.com/Uploads/PDFFiles/04_UVM-Heartbeat.pdf) about this approach.

@rasmus-madsen - as far as the EOT method that you propose using `phase_ready_to_end`, I think this is a cleaner approach as well, I will prepare a PR to update the EOT section of the style guide with this method that should hopefully be out early next week.

weicaiyang, 2020-04-24:
Thanks @udinator 
I will update `cip_lib` to use `phase_ready_to_end` next week

rasmus-madsen, 2020-04-27:
@udinator Great thank you

rasmus-madsen, 2020-05-20:
I think we can close this now?
@weicaiyang do you agree?

weicaiyang, 2020-05-20:
> I think we can close this now?
> @weicaiyang do you agree?

Yes, let me close it. Thanks

closed 2020-05-20 by weicaiyang