lowRISC/style-guides #31: Recommendations around xprop in simulations

GregAC, 2020-03-24:
The style guide currently recommends the following style for case statements

```systemverilog
alway_comb begin
  unique case(in[3:0])
    4'b0001: out = 4'hE;
    4'b0110: out = 4'hA;
    4'b1000: out = 4'h6;
    default: out = 4'h0;
  end
end
```

Under standard system verilog semantics this blocks x propagation but due to the `default:` label we won't see any unexpected values in the output.

However if you turn on xprop in a simulator an `x` value on `in` results in an `x` on out (the simulation sees that `out` could change depending upon the value of `in` so as `in` is `x` it assigns `x` to `out` this is a change to the standard system verilog semantics).

This causes issues when `ASSERT_KNOWN` is used for values that are downstream of such a case statement (i.e. either `out` directly is used in an `ASSERT_KNOWN` or some signal derived from `out` is). When xprop is enabled the `ASSERT_KNOWN` is triggered.

A practical example of this can be found in the Ibex RTL, the `alu_op_a_mux_sel` must always be known:

https://github.com/lowRISC/ibex/blob/2c198383a363c6c8512d9e56fbee126fa74bbf04/rtl/ibex_id_stage.sv#L855

However it's generated via the (style-guide) compliant always_comb statement inside ibex_decoder:

https://github.com/lowRISC/ibex/blob/2c198383a363c6c8512d9e56fbee126fa74bbf04/rtl/ibex_decoder.sv#L543-L791

The decoder itself is fed directly with the instruction flop which will start the simulation as `x`. So when xprop is enabled the `ASSERT_KNOWN` for `alu_op_a_mux_sel` fails.

I can see a few possible solutions:

1. Declare that due to the rules in our style guide (with hopefully some lint to back them up!) that xprop is not a useful feature and shouldn't be used in simulations

2. Explain the xprop issue in the style guide so readers are aware. This may mean many of our unqualified `ASSERT_KNOWN` uses will require an enable term, we may want a new macro `ASSERT_KNOWN_IF` or similar for providing an enable term.

3. Alter the RTL so such signals get a suitable valid factored in so the x that gets propagated via the `case` or `if`/`else if`/`else` statements is squashed.

I don't think 3. is a good idea, so this leaves us with 1. or 2. I'd tend towards 1. but there may be good reasons this is a bad idea I haven't yet realised.

GregAC, 2020-03-24:
@udinator @sriyerg @msfschaffner @sjgitty any thoughts?

rswarbrick, 2020-03-24:
For what it's worth, I prefer (2): then you're explicitly documenting in the (non-synthesized) code why you believe that the RTL isn't buggy.

tomroberts-lowrisc, 2020-03-24:
I'd also go for 1 or 2. As @rswarbrick said, if we don't care that a signal isn't X at certain times, we should probably document it.

vogelpi, 2020-03-25:
Thanks @GregAC for digging into this. I think we should go for Option 2 and document each case appropriately.

GregAC, 2020-03-25:
Ok I think I can be convinced for 2.

I'll put together a PR for an `ASSERT_KNOWN` with an enable term and a PR for the style guide to flag up the issues.

weicaiyang, 2020-03-26:
sorry, late to the party. 
Looks like we prefer to gate on design side in recent xprop issues as below. I'm not sure if the cost to do so is very high for this case.
https://github.com/lowRISC/opentitan/pull/1732
https://github.com/lowRISC/opentitan/pull/1033

@msfschaffner @eunchan @tjaychen any thought?

sriyerg, 2020-03-28:
Me as well (late to the party). 
Without X-prop being enabled, the simulators use an overly optimistic model for treating Xs on mux select lines (sanitizes them to 0), to speed up the simulation. This has been known to mismatch the real / synthesized design behavior and hence potentially mask bugs that are uncovered much later when running gate level sims (if they are even run at all). 

This is elaborated on a bit more in our [DV methodology](https://docs.opentitan.org/doc/ug/dv_methodology) doc (scroll down to xprop). 

(1) IMO is not an option for out project.

+ @cdgori

udinator, 2020-03-28:
Late to the party as well.
I think I am leaning towards option 2, as long as the relevant issues are documented well enough.
I agree with Sri that option 1 should not really be considered (at least from the DV perspective).

cdgori, 2020-03-30:
I would definitely say (2) is preferred to (1), but I have definitely worked on projects where (3) was done as well.

(I thought we had the conversation some ways back about fully enumerating the case and then having the default propagate an X?  In the example it would mean having 16+1 case items, the 3 existing, 13 new ones for the functional default, and then the default that propagates the X.  Even if we did that, we would still need to use something like `ASSERT_KNOWN_IF` to not fire the assertions too aggressively, since they should only trigger when an X value is actually "consumed" by the downstream logic.)

GregAC, 2020-03-30:
I'm currently in the process of adapting Ibex RTL so (2) can work.

I have all of the required RTL changes ready locally, I'm just awaiting the conclusion of the discussion around my prim_assert changes: https://github.com/lowRISC/opentitan/pull/1836. At which point I'll do the PR against Ibex which can then be vendored in to Open Titan.

I'm generally averse to (3) because it can cause timing issues. However there is the possibility of going through the `ASSERT_KNOWN` and `ASSERT_KNOWN_IF` macros and manually adding the enables as gating terms in RTL. Whilst we may not want this in general Ibex (and parameterising it or using a `define would be very messy) it could be an option for a special 'hardened' Ibex.

msfschaffner, 2020-03-30:
Hello and sorry for being late to the party as well.

I think the TL-UL case (lowRISC/opentitan#1732) was somewhat special in the sense that SRAM was actually forwarding illegal data that got consumed and stored in FIFO flops down the road, and we wanted to prevent that from happening. hence we agreed to zeroe them out in RTL.

In this case however, I think I would lean towards 2). Internally, we are using a macro called `` `ASSERT_KNOWN_VALID(name, qualifier, signal)`` for exactly this case.