lowRISC/style-guides #39: [SV] Placement of closing parentheses in initializer lists

imphil, 2020-07-03:
Our style guide currently has this example:

```systemverilog
assign structure = '{
    src: src,
    dest: dest,
    default: '0};
```

Note the placement of the closing `};`.

In the wild, we have a fair bit of code which uses the following style:

```systemverilog
assign structure = '{
    src: src,
    dest: dest,
    default: '0
};
```

(e.g. https://github.com/lowRISC/opentitan/blob/a4a9e4013612607c68086ce4534a14206928f3a6/hw/ip/prim/rtl/prim_keccak.sv#L191-L198)

Currently our style guide doesn't have language (beyond that example) mandating one use over the other. Which option do we prefer?

imphil, 2020-07-03:
This issue should give guidance to verible formatter developers as to which format is expected. See https://github.com/google/verible/issues/352.

asb, 2020-07-04:
I think in general the SV style guide has made choices that should seem familiar to people used to common C/C++ style guides like Google or LLVM's. On that basis, the second option seems the preferred option.

Disclaimer: It also matches my personal preference, so perhaps I'm just backfilling a justification :)

sjgitty, 2020-07-09:
I prefer the second form (closed `}` on its own line) but have been guilty of the first at times. I can work up a quick PR to augment the SG if we're in agreement.

PR #40 (2020-07-20): [vsg] add guidance on closing braces

Resolves #39 

Add clarity on termination characters.
closed 2020-07-20 by sjgitty