lowRISC/style-guides #41: Scope of this style guide and questions

hirooih, 2020-08-20:
Hi,

I've been working on SystemVerilog Style Guide for my company with my colleagues.
I found your style guide and it is very well written.  The goal of your stile guide and ours are very close, because we are also referring to books and papers by Stuart Sutherland, Don Mills, and Clifford Cumming as your Style Guide is.

I have some feedbacks and some proposals on your style guide.
Is your style guide only for lowRISC project?
I think your style guide can be applied to wide range of projects.  If you don't mind I'd like to send pull requests to you.

Here is my fixes ready to be push.
https://github.com/lowRISC/style-guides/compare/master...hirooih:master

I have some other proposals.  Some of them need Q&A with you. Some needs discussion on a separate issue ticket.
Before going to further let me know your thoughts.

Thank you sharing your great style guide.

imphil, 2020-08-20:
Hi @hirooih, thanks for your interest in this style guide. Even though the primary user of this style guide are lowRISC projects, you're certainly welcome to use it for any other code as well.

We are also happy to accept pull requests and changes to the style guide. Note, however, that some topics are rather subjective and we might not be able to reach agreement -- in this case, our internal use cases win :-) Either way, I'm looking forward to your pull requests. A quick look at the diff you linked seems like many changes can go in without much disagreement.

hirooih, 2020-08-20:
@imphil Thank you for your prompt reply.

> Note, however, that some topics are rather subjective and we might not be able to reach agreement -- in this case, our internal use cases win :-)

I understand.

I've sent my pull request #42.  Please review it.

***

Let me ask some questions on this ticket to make other proposals.
Line numbers are ones on my local file which I pushed now.  They may be slightly different from yours, sorry.

#### Case items

L672

> The `default` case item must include a colon.

I don't understand the intention of this sentence.  The `default` case item without colon is just a syntax error.

#### Top-level `parameter` declarations

L1385

> Top-level `parameter` declarations are preferred over `` `define`` globals.

L1548

>  For Verilog-2001 compatible code, top-level parameters are not supported and `` `define`` macros must be used instead.

What do you mean by  "top-level parameters"?
Are they parameters defined in a common package?

#### Catching errors where invalid values are consumed

I don't understand the first sample code.

L1906

> // internal logic which generates reg_addr/reg_wr_en reg_en_addr will never
> // be X but must be ignored if reg_wr_en == 0

reg_en_addr is not defined.
This must  be the follows, but not sure...

> // internal logic which generates reg_addr/reg_wr_en
> // reg_addr can be X and must be ignored if reg_wr_en == 0

Thanks

hirooih, 2020-09-16:
Hi,

I'd like to make next clarification proposals.
Could anyone answer to the questions above?

Thanks.