lowRISC/style-guides #55: Guidance on whether to add space between SV keyword 'wait' and parenthesis

sriyerg, 2021-05-24:
Our current style guide does not appear to have a recommendation for whether we should add a space after `wait` and the opening parenthesis. It will be useful to settle on one so that Verible knows what to do.

Reason for adding space between wait and parenthesis:
- By following the [Spaces around Keywords](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#space-around-keywords) rule, the current rule set is simple - there is no need for exceptions

Reason for not adding space:
- It's an event control statement that behaves very much like a task, so it naturally aligns better with [Function and Task Calls](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#function-and-task-calls) rule.
- Our current code base currently follows this style, so less churn

Interestingly, the LRM also has a mixed usage of both styles. Though, it is of course not the best place to look for style recommendations.

Thanks 
Sri

matutem, 2021-05-24:
I think the current guidelines give a clear recommendation: https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#space-around-keywords recommends adding a space. Simple guidelines are easier to adopt, and with an auto-formatter these issues stop being a concern.

asb, 2021-05-24:
Just as a point of comparison, I think most C++ codebases treat function-like keywords (alignof, decltype, sizeof, const_cast, ..., etc) differently to `for` and `while`, and don't have a spae after the keyword. Although this isn't an explicitly stated rule (unless I missed it), you can see this in the examples throughout the [Google C++ styleguide](https://google.github.io/styleguide/cppguide.html).

rswarbrick, 2021-05-25:
Personally, I'm with @matutem. Note that the full syntax for a wait statement is something like this: `wait (foo) bar;`. Using `wait (foo);` is just a degenerate case (equivalent to doing something like `while (foo());` instead of `while (foo()) {}` in C). When you look at it like that, the construct looks a lot like an `if` statement (`if (foo) bar;`), so should "obviously" have the spaces.

There's also the `wait fork;` syntax, which doesn't have the parens.

My preference: Keep the existing recommendation.

closed 2022-10-31 by sriyerg