lowRISC/style-guides #65: Plagarism of the Google Verilog Style Guide

jonmayer, 2022-10-29:
Hi LowRISC team,

My name is Jonathan Mayer.  I no longer work at Google, but while I did I was the principal author Google's internal Verilog Style Guide.  Especially going back in the change history -- I notice that you seem to have taken that style guide, made some minor changes to it, and published it as your own original work.  While I appreciate that the style guide has finally seen light of day, that sort of seems like plagarism to me? 

Has this document been released by Google under creative commons in some way that I'm not aware of?  If so, why has the attribution of authorship been removed?

The best situation would be for Google to finally publish their original version of this style guide, and for you to properly attribute the people who worked on that document as the original authors.  Can you please endeavor to make this so?

Thank you,

- Jonathan.

hlry, 2022-10-31:
Hi Jonathan, please be careful throwing around legal-sounding accusations. There has been no foul play here; Google is on the board of lowRISC and collaborates closely with lowRISC developers. We can correspond privately if you have concerns about attribution, but as you note, you are not authorized to represent Google here. Our mutual connections will share my google.com email address with you so I can make sure that I understand your concern appropriately. - Hilary (Google OSS Licensing & lowRISC legal committee member)

jonmayer, 2022-10-31:
I understand.

closed 2022-10-31 by jonmayer