lowRISC/style-guides #80: Curly braces in coverpoints must close with the first character of the coverpoint name

antmarzam, 2025-05-09:
This suggestion is followed in Opentitan

Good Example:
```
Covergroup cg;
  Bit x;
  Cp_name: coverpoint x {
    Bins zero = {0};
    Bins one  = {1};
  }
endgroup
```
Bad Examples:
```
Covergroup cg;
  Bit x;
  Cp_name: coverpoint x {
                         Bins zero = {0};
                         Bins one  = {1};
                        }
Endgroup

Covergroup cg;
  Bit x;
  Cp_name: coverpoint x {Bins zero = {0};
                         Bins one  = {1};
                        }
Endgroup

Covergroup cg;
  Bit x;
  Cp_name: coverpoint x {Bins zero = {0};
                         Bins one  = {1};}
Endgroup

```

marnovandermaas, 2025-05-09:
What about?
```
Covergroup cg;
  Bit x;
  Cp_name: coverpoint x
  {
    Bins zero = {0};
    Bins one  = {1};
  }
endgroup
```

antmarzam, 2025-05-09:
Is not what's followed in OpenTitan I think. But happy to include if others agree!