lowRISC/style-guides #81: Check the return value of the predict method

antmarzam, 2025-05-09:
Do not cast the uvm_reg predict method to void, instead check the return value.

Argument:
The predict method may fail depending on whether it uses UVM_PREDICT_READ, UVM_PREDICT_WRITE or UVM_PREDICT_DIRECT, and ignoring the return value effectively hides whether the prediction was successful or not.

marnovandermaas, 2025-05-09:
Can we expand this rule to apply to all UVM functions?

rswarbrick, 2025-05-12:
I think I'd be inclined to expand it too. Are there any UVM functions with return values that we don't care about? (I suspect not, but maybe it's worth explicitly asking)

Also, can we easily create a list of UVM functions with non-void return values? If it's short, it might be worth including as a supplementary note on the guideline?