lowRISC/style-guides #83: One declaration/statement per-line [Lexical suggestion]

antmarzam, 2025-05-09:
Reason: helps with code-annotation when debugging code without having to hack the source directly.

marnovandermaas, 2025-05-09:
I like this suggestion, but am worried about the code churn this would require.

rswarbrick, 2025-05-12:
I agree with @marnovandermaas.

Is it easy to get a feel for how many violations there are in OpenTitan? I did a really stupid search with
```
git grep '^[^/]*;\s*[^ /,)\\]' '*.sv' | grep -v 'for\s*(' | grep -v constraint | grep -v bins | grep -v while | grep -v import | grep -v ':\s*`' 
```
and found 244 lines. (The last `grep` call drops some macro invocations).

If we wanted to specify something like this, I think we'd need to be a bit more specific about what we're telling people to avoid.

rswarbrick, 2025-05-12:
One reason to avoid multiple declarations on a line might be that it's easy to forget to add comments to some of the variables.

nbdd0121, 2025-05-12:
What about `_q`, `_d`? I think they should be allowed in the same line.

antmarzam, 2025-05-12:
> What about `_q`, `_d`? I think they should be allowed in the same line.

I think that's fine, this is only targeting DV guidelines