lowRISC/style-guides #85: Don’t add virtual to overridden UVM methods [lexical guidelines]

antmarzam, 2025-05-09:
Reason: They don’t have an effect (methods are still virtual), they just clutter the code.

marnovandermaas, 2025-05-09:
What if you have a grandparent, parent and child class. If you have a virtual function in the grandparent class but your rule says that you may not mark it as virtual in the parent class. Then your child class can override the same function but you would have to go back to the grandparent class to see whether it is virtual or not. I think it would be more clear if we do mark the function in the parent class as virtual even if it is redundant.

rswarbrick, 2025-05-12:
Hmm, I think I probably agree with @antmarzam. Defining a method with a matching name that *isn't* supposed to override a virtual one is an excellent way to confuse readers! I don't think it's worth explicitly worrying about making that easy...